chipsalliance/python-fpga-interchange #8: Need canonical storage solution to ensure binary diffablity

litghost, 2021-01-21:
Currently there are two sources of instability in the FPGA interchange capnp files:
 - strList order is unstable
 - Inconsistent compression output

This issue is to track adding logic to remove instability in the FPGA interchange capnp files.  The strList order can be fixed by sorting the strList and re-numbering the ids.  The inconsistent compression output can be solved by ensuring that the compress level between tools is consistent, and using a compression library that generates stable results.

mithro, 2021-01-21:
I don't think diffability is the major goal. More making sure that two files which have identical data contents will be identical.

IE If the diff tool returns that two files are identical, then the canonical storage version of the files should have identical checksums.

litghost, 2021-01-21:
> I don't think diffability is the major goal. More making sure that two files which have identical data contents will be identical.
> 
> IE If the diff tool returns that two files are identical, then the canonical storage version of the files should have identical checksums.

Agreed, I'll update the description.