chipsalliance/python-fpga-interchange #19: Problems with rapidyaml requirement

ajelinski, 2021-02-15:
There are two problems with `rapidyaml`:
1. While installing this package with `pip install git+https://github.com/symbiflow/python-fpga-interchange.git`, there is an error that rapidyaml can't be found which is true since it's in the `setup.py` in `install_requires` but it's not a PIP package.
2. I'd simply solve it by adding `git+https://github.com/litghost/rapidyaml.git@fixup_python_packaging#egg=rapidyaml&subdirectory=api/python` before this package in https://github.com/SymbiFlow/fpga-tool-perf/blob/0888f31ca8855141a7648189167cacc2a6749755/conf/nextpnr/requirements.txt#L4 which should solve the problem of the unavailable `rapidyaml` package, BUT there's an error during its installation. It requires `swig` to be installed in the environment and adding `swig` to the Conda's `environment.yml` doesn't work. I guess it'd be needed to add it in the host OS or which doesn't seem to be the optimal solution.

For now I'll propose to simply lock revision in fpga-tool-perf's `nextpnr/requirements.txt` and `symbiflow/requirements.txt` to the last revision prior to using `rapidyaml`, i.e. https://github.com/SymbiFlow/python-fpga-interchange/commit/e632a5e3e51eea653ff1c73919b2553e5626bb36 

BTW. @litghost I'd add the second point as an issue in https://github.com/litghost/rapidyaml/ but there's no `Issues` tab enabled.

litghost, 2021-02-15:
> 2\. It requires `swig` to be installed in the environment and adding `swig` to the Conda's `environment.yml` doesn't work. I guess it'd be needed to add it in the host OS or which doesn't seem to be the optimal solution.

Adding swig to conda is what was done here, and it appears to have worked:  https://github.com/SymbiFlow/symbiflow-arch-defs/pull/2012

ajelinski, 2021-02-15:
Oh you're right. I've just checked locally that it works. I must've been trying to add `swig` before I found out that also `rapidyaml` should be added separately to `requirements.txt`.

Thanks!

closed 2021-02-15 by ajelinski