chipsalliance/python-fpga-interchange #27: Need a FPGA interchange to FASM generator

litghost, 2021-02-25:
Now that a basic P&R flow has been create that directly consumes FPGA interchange logical and physical netlist, there is a need to generate FASM directly from the FPGA interchange logical and physical netlist.  Ideally this implementation would be able to target multiple architectures.  Initial targets would be Artix7 and Zynq US+, as those are the architecture where there is a complete enough bitstream database and FPGA interchange device database to support those arches.

litghost, 2021-02-25:
This issue will need to grapple with some of the questions raised in https://github.com/SymbiFlow/fpga-interchange-schema/issues/12  Solution to https://github.com/SymbiFlow/fpga-interchange-schema/issues/12 will change requirements on this tool.

acomodi, 2021-04-06:
@litghost I am trying to get an initial working implementation for the FASM generator, initially tailored for 7-series, while trying to maintain it generic where possible.

One thing I stumbled upon is the generation of FASM features corresponding to route-thru pseudo PIPs. The problem is that the physical netlist does not have any information on whether a PIP is a pseudo one, (e.g. route-thru the OLOGIC site when there is no real BEL used). This would be relevant information to attach the correct FASM features when this kind of pseudo PIP is used. Would it make sense to add to the phys netlist schema an `isPseudo` field to the PIPs and emit that as well?

litghost, 2021-04-06:
> Would it make sense to add to the phys netlist schema an `isPseudo` field to the PIPs and emit that as well?

You should be able to consult the `DeviceResource` schema to figure out pseudo pips.  There is still the problem of how to emit the FASM features for the pseudo pips, but for now I'd recommend doing a simple table lookup.  Eventually the `DeviceResource` schema should contain enough information to emit the FASM features by understanding pseudo cell placements, but that is potentially hard, so could be considered out of scope.

acomodi, 2021-04-06:
@litghost All right, I thought about getting information from the DeviceResources and the only initial problem I had was related to run-time, but I think I know what was wrong and I should be able to cut the overhead.

acomodi, 2021-06-16:
An initial version of the FASM generator is already merged and working for both xc7 and nexus. Further improvements and generalizations of the FASM generator can be discussed in follow-up issues. Closing

closed 2021-06-16 by acomodi