chipsalliance/python-fpga-interchange #34: Requiring a NULL tile at (0, 0) is Xilinx-specific

gatecat, 2021-03-17:
https://github.com/SymbiFlow/python-fpga-interchange/blob/master/fpga_interchange/populate_chip_info.py#L1139 assumes that the tile at (0, 0) is NULL.

This works for Xilinx, and can be made to work for Nexus by renaming that tile (because it's equivalent to a NULL tile) but this should be made generic in the future.

litghost, 2021-03-17:
Totally valid!  We can probably detect the "NULL" tile by checking if it has sites or connected routing.

gatecat, 2021-03-17:
I don't think there is a guarantee that there will be a "NULL" tile - I think some iCE40 devices have something in every tile, for example.

litghost, 2021-03-17:
> I don't think there is a guarantee that there will be a "NULL" tile - I think some iCE40 devices have something in every tile, for example.

Weird!  What is in the "corner" of ice40 FPGA's?

litghost, 2021-03-17:
The other option here is to glom the constant source site onto a unique tile in the fabric.  Does the ice40 fabric has at least 1 unique tile in the fabric?

gatecat, 2021-03-17:
> Weird! What is in the "corner" of ice40 FPGA's?

the UltraPlus notionally has hard IP and SPRAM located there

> The other option here is to glom the constant source site onto a unique tile in the fabric. Does the ice40 fabric has at least 1 unique tile in the fabric?

yeah, the (0, 0) corner tile is good enough for this just might not necessarily be empty