chipsalliance/python-fpga-interchange #66: Cleaner nextpnr BBA generation

gatecat, 2021-04-23:
Currently the nextpnr BBA generation code includes a lot of duplication, particularly where lists are concerned. e.g.  patterns like https://github.com/SymbiFlow/python-fpga-interchange/blob/6a80233ca97427c9b1a1cdd16349c3724914d008/fpga_interchange/chip_info.py#L658-L672 crop up a lot

I think a custom decorator might be the nicest solution here, although I don't have enough Python experience personally to comment on how much work it would be to implement this.

Longer term, the structures in nextpnr (https://github.com/YosysHQ/nextpnr/blob/master/fpga_interchange/chipdb.h) and Python should come from the same source to reduce the risk of mismatches, although I don't know which should be definitive.

kowalewskijan, 2021-04-26:
As for me, I think an auto-alignment feature is worth to add too (as a part of this issue) to avoid manual padding for some fields like it is done in [BelInfo](https://github.com/SymbiFlow/python-fpga-interchange/blob/master/fpga_interchange/chip_info.py#L109-L110).

kowalewskijan, 2021-04-26:
> Longer term, the structures in nextpnr (https://github.com/YosysHQ/nextpnr/blob/master/fpga_interchange/chipdb.h) and Python should come from the same source to reduce the risk of mismatches, although I don't know which should be definitive.

I think the header (`chipdb.h`) could be generated as a part of `nextpnr_emit.py` script. Currently in the `Nextpnr` we call the `nextpnr_emit.py` from the cmake during `bba` generation - probably this is the best place to generate the header as we won't re-generate the bba file too frequently.